parallaxinc/Parallax-IDE #209: Deleting file causes phantom new file and Save As dialog

PropGit, 2015-07-28:
__Problem__
  * Make sure there's at least two files in the Project/File View
  * Select one of the files
  * Choose the Delete menu option
  * Answer the warning prompt with _Yes_
  * [The editor blanks out (expected) and toast appears saying delete of file was successful (expected)]
  * Select another file
  * Parallax IDE unexpectedly prompts with "Do you want to save the changes you made to New file?"  Also, even more incorrect, is that the filename field is filled with the name of the file you just selected.

__Solution__
  * Silently skip the "Do you want to save..." prompt if the _new and unsaved_ file in the editor contains only whitespace.

phated, 2015-08-07:
The refactor handles this, but we only check the text content has no length.  If you were to type a space or tab, it would prompt.  Is that not desired behavior?

PropGit, 2015-08-07:
Can you wrap the text content in a trim function so the check would equate the whitespace-only content to no length?  That would achieve the results we want.

phated, 2015-08-07:
Yep, can do.  That's a really quick change and will be fixed with the refactor.

closed 2015-08-07 by 2fast2fourier

PropGit, 2015-08-17:
Verified correct.