parallaxinc/Parallax-IDE #211: Debug Terminal: 10,13 processing filters out too much

PropGit, 2015-07-28:
The processing of LF and CR (individually) and LF/CR or CR/LF (pairs) works fantastically, except for one case... multiple pairs of LF/CR or CR/LF get filtered out.

Just like receiving multiple contiguous individuals LF, LF (or CR, CR) results in two effective new lines (as is properly implemented in Parallax IDE v0.6.0 now), receiving pairs like LF, CR, LF, CR (or CR, LF, CR, LF) should result in two effective new lines, but as-implemented currently. the second (and beyond) pairs get filtered out, resulting in just one effective new line.  This is described by the [Debug Terminal Document](https://docs.google.com/document/d/1u0L0WTVU-jfvN4H56efHdBhNYTbObEX72Bn7Fc2RxwE/edit?usp=sharing) in the Receive Pane's _ASCII 10 (Line Feed) or 13 (New Line)_ section.

The problem is exhibited in the Receive Pane by this contrived test code, between Receive Pane lines "Text: 26" and "Text: 45".

```
'{$STAMP BS2}
'{$PBASIC 2.5}

  PAUSE 1000

  DEBUG "The following should have contiguious numbers with no blank lines in-between", CR
  FOR W0 = 1 to 10
    DEBUG "Text: ", DEC W0, CR
    PAUSE 100
  NEXT

  DEBUG REP CRSRUP\5, REP CRSRRT\3, REP CR\5

  FOR W0 = 11 to 20
    DEBUG "Text: ", DEC W0, LF
    PAUSE 100
  NEXT
  
  DEBUG REP CRSRUP\5, REP CRSRRT\3, REP LF\5

  FOR W0 = 21 to 30
    DEBUG "Text: ", DEC W0, CR, LF
    PAUSE 100
  NEXT

  DEBUG REP CRSRUP\5, REP CRSRRT\3, CR, LF, CR, LF, CR, LF, CR, LF, CR, LF 

  FOR W0 = 31 to 40
    DEBUG "Text: ", DEC W0, LF, CR
    PAUSE 100
  NEXT
  
  DEBUG REP CRSRUP\5, REP CRSRRT\3, LF, CR, LF, CR, LF, CR, LF, CR, LF, CR
  
  FOR W0 = 41 to 50
    DEBUG "Text: ", DEC W0, CR
    PAUSE 100
    DEBUG LF
  NEXT
  
  FOR W0 = 51 to 60
    DEBUG "Text: ", DEC W0, LF
    PAUSE 100
    DEBUG CR
  NEXT
  
  DEBUG "The following should have contiguious numbers with 1 blank line in-between", CR
  FOR W0 = 61 to 70
    DEBUG "Text: ", DEC W0, CR, CR
    PAUSE 100
  NEXT
  
  FOR W0 = 71 to 80
    DEBUG "Text: ", DEC W0, LF, LF
    PAUSE 100
  NEXT

  FOR W0 = 81 to 90
    DEBUG "Text: ", DEC W0, CR, CR, LF
    PAUSE 100
  NEXT

  FOR W0 = 91 to 100
    DEBUG "Text: ", DEC W0, LF, LF, CR
    PAUSE 100
  NEXT
  
  FOR W0 = 101 to 110
    DEBUG "Text: ", DEC W0, CR, CR
    PAUSE 100
    DEBUG LF
  NEXT
  
  FOR W0 = 111 to 120
    DEBUG "Text: ", DEC W0, LF, LF
    PAUSE 100
    DEBUG CR
  NEXT
  
  DEBUG "The following should have contiguious numbers with 2 blank lines in-between", CR
  FOR W0 = 121 to 130
    DEBUG "Text: ", DEC W0, CR, CR, CR
    PAUSE 100
  NEXT
  
  FOR W0 = 131 to 140
    DEBUG "Text: ", DEC W0, LF, LF, LF
    PAUSE 100
  NEXT
```

2fast2fourier, 2015-08-27:
Fixed in bs2-serialport 0.8.0 ([commit](https://github.com/irkenjs/bs2-serialport/commit/6eb9b9f7ca47f1b57f73cc563483c974567db1dc))

The above test results appear correct, all three sections have the correct whitespace.

closed 2015-08-27 by 2fast2fourier

PropGit, 2015-08-30:
Verified fixed!