parallaxinc/Parallax-IDE #213: Debug Terminal: TX/RX - Increase activity visibility

PropGit, 2015-07-28:
To increase communication activity visibility on the TX/RX indicators:
  * Increase the minimum time the TX/RX indicators are _lit_ to about 100 ms; assuming it's currently limited to a minimum time of about 50 ms as requested in [Issue # 78](https://github.com/parallaxinc/Parallax-IDE/issues/78#issuecomment-110996974).
  * After the above time expires (TX/RX lit), force an _unlit_ moment of no less than about 25 ms (before lighting again) so that the TX/RX indicators have a visible _blink_ even when a continuous stream of data is arriving at a high rate of speed.

2fast2fourier, 2015-10-16:
I've submitted this as PR #302, but with a minimum unlit time of 50ms. When testing the flashing on a fast loop, 50ms looked a lot cleaner than a 25ms period.

closed 2015-10-16 by phated

PropGit, 2015-10-16:
@2fast2fourier - Thank you!

PropGit, 2015-10-21:
Verified fixed on Rx, but Tx seems to not blink at all.  Opened a new issue #310.

PR #302 (2015-10-16): Add rx/tx minimum unlit time, adjust minimum lit time
#### What's this PR do?
This PR adds a 50ms minimum unlit time to the RX/TX flash cycle, and increases the minimum lit time to 100ms.
#### What are the important parts of the code?
This adds `RX_CLEAR_TIMEOUT`/`TX_CLEAR_TIMEOUT` events in redux. This extends the RX_ON/RX_OFF chain to include `RX_CLEAR_TIMEOUT`.
#### Is any other information necessary to understand this?
Issue #213 suggested a minimum 25ms unlit time, but in testing 50ms looked a bit cleaner.
#### What are the relevant tickets?
Closes #213