parallaxinc/Parallax-IDE #225: Question: some auto download behavior not defined?

phated, 2015-08-06:
@PropGit I thought up a scenario that I don't think we covered in the 4 items for the automatic download feature that was implemented.  See below.

**Scenario:** User has 2 BS2s and a BS2e attached to their computer.  Their source code has a BS2 directive.  They attempt to download.  They are prompted to choose a device due to having 2 BS2s attached but they select the BS2e.

**Questions:** What should be the behavior of them selecting the BS2e?  Should it rewrite the source code, like in the "no match" scenario?

PropGit, 2015-08-06:
@phated Exactly.  During the scan for BASIC Stamps, if a single target can not be automatically reasoned, the app is left to ask the user, "Which one?"  If the answer the user gives doesn't match the module targeted by the code, the software must try to modify the code and recompile for that "answer" and then automatically download to that selected port if the recompile was successful.

phated, 2015-08-07:
Thanks, while doing the refactor, I found the code used around this case was fragile and was able to make it better/more robust thanks to your feedback.

PropGit, 2015-08-07:
Great find!  I appreciate it!

closed 2015-08-07 by 2fast2fourier