parallaxinc/Parallax-IDE #238: Can't test syncFileSystem; it seems to not be working

PropGit, 2015-08-19:
syncFileSystem has never worked for me.  ie: I've created many source files in Parallax IDE on my computer while logged into a particular account in Chrome and those files never show up in Parallax IDE on another computer while logged in to the same account.

I've specifically retested this again today between my main development machine and my laptop (both Win 7 Pro 64-bit) logged into Chrome using the same account.

My main machine's Parallax IDE (v0.6.0 and also v0.7.0) has files I've created within them in the default project "/new-project".

On my laptop,
 * Parallax IDE v0.6.0 automatically shows up (since it was added from the web store while on my main machine)
 * None of the source code I created in v0.6.0 on my desktop show up on my laptop... Parallax IDE opens with a blank "/new-project"
 * I can open Google Drive > Recent and can see the source code I expected to show up
 * Interestingly, I sat there staring at Parallax IDE for a while upon the first session, and after about 5 seconds, for a brief moment I saw a file appear in the /new-project list, but then swiftly disappeared (it only stayed for about 1/3 second... too fast for me to read what it was titled).

So my questions are:
  * Does this still work for you?
  * What's different about my setup?

PropGit, 2015-08-19:
NOTE: For v0.6.0, I verified that both systems showed the extension with the same ID.  They do- exactly the same:

[image]

PropGit, 2015-09-01:
Implement Issues #239, #240, and #241 to assist with addressing this issue.

urbantumbleweed, 2015-09-14:
This issue is at a point where internal decisions need to be made within Parallax.  The discussion will continue when desired interactions are clearly defined and can be translated into smaller issues.

PropGit, 2015-09-28:
__Update__
@phated and @urbantumbleweed - I've been figuring out cases where our current implementation does something destructive or unexpected (from a user's point of view, not necessarily from ours).  It's difficult to realize this because there are so many possibilities that are hard to wrap my mind around and part of the nature seems somewhat indeterministic.  I finally realized I need to make a chart of steps to setup, perform, and see results - that has helped, but I feel I still have further to go before seeing the whole picture.  

I hope to wrap up the cases and make decisions in the next day or so.  From my current perspective, I think the solution will be a multifaceted one; some parts possible soon, others only possible later after much more work is in place.

Knowledge from issues #279 and #280 will help me feel out the boundaries of what's possible and practical.