parallaxinc/Parallax-IDE #239: Create account name view

PropGit, 2015-08-19:
Because people could use multiple Google accounts, or stay logged out, we need a way for Parallax IDE to display the Google account being used.

I like how Chrome does this:
[image]
nice and tidy.  Let's do something equally concise; exactly the same if possible.

It should display the user's name on their account if they are logged in.
It should display "log in" and use a different background color (to catch the eye) when they are not logged in.

PropGit, 2015-09-17:
Per discussion: There _may_ be some way to retrieve user's account name or email address... or there may not be a way.

phated, 2015-09-17:
It seems that another permission was needed to get the email address with the `chrome.identity.getProfileUserInfo` API (more information at http://stackoverflow.com/a/25547737).

However, that stack overflow comment has a reply that says it always returns the first account if multiple accounts are signed in to chrome.  I haven't tested to see if that still is the case (the thread is from 2014).

PropGit, 2015-09-17:
@phated - Awesome!  Looks like they've updated the documentation to say that the `identity.email` permission must be set.

It also says: The information returned is available offline, and __it only applies to the primary account for the profile__.

What does _primary account_ mean for us?  I'd think that's what we want.

phated, 2015-09-17:
I think primary account means "first account added" not current account, unfortunately.  I'm just basing that on the stack overflow answer linked and haven't investigated that assumption.