parallaxinc/Parallax-IDE #247: Hide the "Don't Save" button when discarding a file isn't a desired behavior

phated, 2015-08-29:
Currently, if a file is a brand new (unsaved) file, the Save/Save As prompt has a "Don't Save" button.  This is desired when a user attempts to change from a new file to a previously saved file and is prompted to save.  However, this is incorrect UI behavior when the user is just using the Save/Save As functionality of the editor.  We need to hide the "Don't Save" button when we are not in the "Changing File" mode.

PropGit, 2015-08-29:
NOTE: This is a new behavior (bug) in v0.7.1; it did not behave this way in v0.7.0 as you can see about 20 seconds into [this video](https://drive.google.com/file/d/0B0tev9Y3WaL4QTBFWG5SVUtwMVk/view?usp=sharing) that was recorded for issues #235.

closed 2015-09-14 by phated

PropGit, 2015-09-16:
Verified fixed.  However, this has led to a side effect noted in issue

commit c1b4f43e (2015-09-14): added a another case to overlays to account hide the 'don't save' button. closes #247
PR #263 (2015-09-14): Hide the 'don't save' button when in change mode. closes #247
#### What's this PR do?
It removes the "don't save" button from the save overlay when in change_file mode.
#### What are the important parts of the code?
In handlers the `showSaveOverlay` function now takes a boolean flag to hide/show button.  Also, in the overlays.js plugin there is a new case that passes the boolean as props.
#### How should this be tested by the reviewer?
`git pull origin remove-dont-save-button-on-change-file && npm i && npm run build`
Then launch the app and create new files and then either save or navigate awa [... cut]
commit 5c339d26 (2015-09-14): Merge pull request #263 from parallaxinc/remove-dont-save-button-on-change-file

Hide the 'don't save' button when in change mode. closes #247