parallaxinc/Parallax-IDE #253: New files get default directives

PropGit, 2015-09-01:
We currently don't have Stamp and PBASIC directive quick-access buttons - we need to ease the pain of users creating new files.

For a temporary time, we want all new files to automatically have the default Stamp and PBASIC directives inserted into them.
```
'{$STAMP BS2}
'{$PBASIC 2.5}
```

phated, 2015-09-01:
Note: this will cause the user to __always__ be prompted to save a new file because it now has text in it.

PropGit, 2015-09-01:
Understood; we accept those results as having the directives are normally what the user wants as he/she is starting to write new code anyway.

This is a temporary measure that will be replaced with a different, although similar, method of swiftly getting the directives in-place.  We may or may not choose to address the non-blank-but-still-disposable source issue in a nicer way in the future.

PropGit, 2015-09-09:
UPDATE: I updated the first post's code to include two blank lines.

Is it possible to place the cursor at the start of the 4th line (2nd blank line) when, and only when, this text is placed in the edit by the New operation?

closed 2015-09-14 by phated

PropGit, 2015-09-16:
Verified done.

PR #266 (2015-09-14): add directive builder function and modify newFile handler
#### What's this PR do?
Adds functionality to automatically add directives into new files.
#### What are the important parts of the code?
Adds a directive builder function that takes a board type and language.  When a newFIle is created the directive builder sets the contents of the temp file and the value in the document to the directive.  Once the directive is inserted and the editor is focused, there is a call to move the cursor to the end of the document. 
#### How should this be tested by the reviewer?
Download branch, launch edito [... cut]