parallaxinc/Parallax-IDE #270: application context architecture

urbantumbleweed, 2015-09-16:
There are a number of keyboard shortcuts that should work differently depending on the current view state of the application.  The concept of context has not yet been built into the application and will need to be prior to being able to resolve issues that depend on the the feature.

Examples of issues that depend on context are: #104 and #105.

PropGit, 2015-09-17:
Context also affects mouse responses and touch gestures (when on touch-interface devices).

PropGit, 2015-09-17:
There may also be logical groups of keys that can be defined and those groups have the same meaning in different contexts, while no meaning in other contexts.

PropGit, 2015-09-17:
Brainstorming...  

__Key Groups:__
  * __Text__
    * Keys that emit glyphs by themselves (0x20..0x7E, 0xA0..0xFE, and most Unicode glyphs)
    * Alt-# combinations
    * _FUTURE_: Propeller support will add additional glyph codes which are emitted for keys 0x02..0x07, 0x14..0x1F, 0x7F..0x9F
  * __Text Tab__
    * Tab (0x08)
  * __Text Enter__
    * Enter (0x13)
  * __Text Edit and Navigation__
    * Left, Right, Up, Down, Ctrl-Left, Ctrl-Right, Shift-Left, Shift-Right, Ctrl-Shift-Left, Ctrl-Shift-Right, Alt-Left, Alt-Right, Shift-Up, Shift-Down, Ctrl-A, Ctrl-C, Ctrl-X, Ctrl-V, Ctrl-Up, Ctrl-Down, Ctrl-Home, Cmd-Up, Ctrl-Search-Left, Ctrl-End, Cmd-Down, Ctrl-Search-Right, Home, End, Shift-Home, Shift-End, Ctrl-Shift-Home, Ctrl-Shift-End, PageUp, PageDn, Shift-PageUp, Shift-PageDn, Bksp
  * __Source Entry__
    * Includes __Text__ plus __Text Enter__ plus special Tab, Shift-Tab (emits or removes groups of spaces)
  * __Source Edit and Navigation__
    * Includes __Text Edit and Navigation__ plus Ctrl-D, Ctrl-Z, Ctrl-Shift-Z, Ctrl-F, Ctrl-G, Ctrl-Shift-G, F3, Shift-F3, Ins, Del, Ctrl-Tab, Ctrl-Shift-Tab
  * __Operation__
    * Crl-O, Ctrl-N, Ctrl-S, Ctrl-Shift-S, Ctrl-P, Ctrl-I, F6, Search-6, Ctrl-T, F7, Search-7, Ctrl-R, F9, Search-9, F1, Search-1
    * _FUTURE_: F8, Search-8, Ctrl-M, plus more future Propeller support functions
  * __App Navigation__
    * Ctrl-< (TBD), Ctrl-> (TBD), Esc
  * __Prompt Navigation__
    * Tab, Shift-Tab, Esc, Enter

__Contexts:__
[Only the Key Groups listed are active in each context]
  * __Editor__ - Focus is on the source editor (most common)
    * __Source Entry__ keys affect source code directly
    * __Source Edit and Navigation__ keys move cursor around in source code, affect view port, and switch to other source
    * __Operation__ keys trigger app functions (New, Save, Download, Identify, etc.)
    * __App Navigation__ keys switch focus to other contexts
  * __Terminal__ - Focus is on the Transmit or Receive Pane of Debug Terminal
    * __Text__, __Text Tab__, and __Text Enter__ - If focus is on Receive Pane, focus switches to Transmit Pane and emits text characters
    * __Text Edit and Navigation__ - moves around in text with keyboard
    * __Operation__
    * __App Navigation__
  * __Terminal Status Panel__
    * (TBD)
    * __Operation__
    * __App Navigation__
  * __Project/File View__
    * (TBD)
    * __App Navigation__
  * __Memory Map__
    * (TBD)
    * __App Navigation__
  * __Prompt__
    * __Prompt Navigation__

urbantumbleweed, 2015-09-28:
@PropGit After much thought and discussion on this issue a concept has emerged that we are proving internally. Here is a summary of the thought processes thus far.

The current state of the application does not include the concept of context. This means that event handlers are registered at the root level of the application and are triggered universally during a user's session. For this reason, pressing the 'tab' key while viewing a modal would still affect the text editor that is opened behind the modal while it may or may not toggle through the modal button options.

Additionally, issue #276 highlights the need for us to maintain a history of context state.  For example, while canceling on an "overwrite" prompt, we are taken back to a save prompt but not necessarily the save prompt that led up to the "overwrite" prompt; this happens because we are not using history to return to the previous dialog but instead have dom elements wired to handlers and therefore cannot determine which context we came from.

So as we got to thinking about this, our first approach was to create a context stack with context objects including both type and metadata (used for populating views and handlers).  The idea here is that context state could be added to and popped from the stack, resolving the history issue of the "overwrite" dialog.  This seemed to solve part of our problem, but it did not provide us with an answer on the need to change the meaning of a specific event (i.e tabbing) in different contexts.

All in all we considered a couple different ideas, all of which required us to architect a system to manage context state, track view hierarchy, and manage history.  While the initial ideas came close in terms of resolving the current problem, it would be at the expense of complicating the code-base and reducing maintainability long-term. So, after much deliberation we came to realize that this problem relates very closely to that of webpages and could be solved through giving each "context" a separate route.  However, being in a chrome application removes the APIs we would use to do routing.

While investigating our options we considered a well established open-source library called 'React-Router'.  We are very optimistic about this direction.  By default React Router uses the html history api, but it has the ability to manage its own state history object in memory. Furthermore, it simplifies our idea of contexts by using 'routes', which in our case are simply various states of the application. We can use the routes to determine what set of functionality is available to the current view.

After taking a look throughout the weekenend and testing the library within the confines of a packaged Chrome application, we have confirmed that this is promising avenue to pursue. This is a major undertaking and will not be implemented by the end of this sprint.  It is clear there are many things that could change regarding the way we construct our views and handle events.  We will be in a better position to discuss this by the demo, but will likely need another sprint cycle to complete the implementation.

PropGit, 2015-09-28:
@urbantumbleweed - Awesome work guys!  Sounds like you really dove into this as a team and explored things deeply.  I like the sound of the solution you are headed towards.  Thank you for all the details.

> It is clear there are many things that could change regarding the way we construct our views and handle events.

This makes me nervous.  Maybe you can fill me in on what you mean here in our meeting.

Regarding using React-Router routes to keep context history, something occurred to me to be careful about.  (I'm not sure it will be a problem, but thought I should pose the question).  Will this have any effect on inter-session behavior?
  * For example, right now if you do a Save As and enter an already existing filename, the app presents the Overwrite prompt... but if the app is closed at that moment (without acknowledging the prompt), restarting brings it back in a different state (the overwrite prompt is not there).
What I'd want you to be thinking of and protecting against are situations like this that would perhaps unintentionally preserve state and context across abortive actions (app closes from user or system action, or some other system or app error that abruptly aborts a call chain of some sort, whether real or logical).

urbantumbleweed, 2015-09-29:
@propGit: We are going to discuss this further on our call Wednesday when we can the discuss technical implications in regards to changing views.

One thing to keep in mind about React-Router state is that the history object is only in memory and will behave similarly to how undo-redo history works on documents.

We will discuss further and if needed on our call.

monteslu, 2015-10-16:
update:

I've been working on changes including replacing holovisor dependency with a new layout view.
This view is the main Route element "/" of an instance of react-router.  Its child elements are the overlays which get rendered when we call pushState() of the overlay we wish to see.  This allows us to move backwards to a previous context when we do something such as canceling a nested overlay dialog.

This work also partially replaces the skrim dependency, however we are still going to be using that plugin for toasts for time being.

Major pieces left are updates to the overlay actions to use pushState as well as a remapping of some global key handlers to their specific contexts for the dialogs.

PropGit, 2015-10-26:
@monteslu - I updated the [Key Groups and Context definition post](https://github.com/parallaxinc/Parallax-IDE/issues/270#issuecomment-141197000).

PR #314 (2015-10-30): Router
#### What's this PR do?
Refactors internals that were dependent on Irken to use react-router and removes several plugins. It also changes the render handling of overlays to further support context based on dom and routes.
#### Where should the reviewer start?
Start in client.js.  This file is what creates the router instance
#### How should this be manually tested?
pull down branch, build and run through application
#### What are the relevant tickets?
closes #270
closed 2015-10-30 by phated