parallaxinc/Parallax-IDE #272: What keyboard shortcuts is CodeMirror supporting?

PropGit, 2015-09-17:
Is there a list of keyboard shortcuts you can provide that CodeMirror is supporting by default (in Parallax IDE)?

urbantumbleweed, 2015-09-17:
This is something we can look into.  The application did modify some of the default behavior so to provide a complete answer we will need to look at that as well.

PropGit, 2015-09-17:
Yes, perfect.  Thank you.

urbantumbleweed, 2015-09-23:
Jeff,

Here are the key-mappings that are enabled by default in CodeMirror.  Included are key-mappings for both 'pc' and 'mac'.  Below the first list are CodeMirror key-bindings that have been disabled replaced custom mappings in Parallax through our key-mapping library, Iggins.

{ Left: 'goCharLeft',
  Right: 'goCharRight',
  Up: 'goLineUp',
  Down: 'goLineDown',
  End: 'goLineEnd',
  Home: 'goLineStartSmart',
  PageUp: 'goPageUp',
  PageDown: 'goPageDown',
  Delete: 'delCharAfter',
  Backspace: 'delCharBefore',
  'Shift-Backspace': 'delCharBefore',
  Tab: 'defaultTab',
  'Shift-Tab': 'indentAuto',
  Enter: 'newlineAndIndent',
  Insert: 'toggleOverwrite',
  Esc: 'singleSelection',
  'Ctrl-A': 'selectAll',
  'Ctrl-D': 'deleteLine',
  'Ctrl-Z': 'undo',
  'Shift-Ctrl-Z': 'redo',
  'Ctrl-Y': 'redo',
  'Ctrl-Home': 'goDocStart',
  'Ctrl-End': 'goDocEnd',
  'Ctrl-Up': 'goDocStart',
  'Ctrl-Down': 'goDocEnd',
  'Ctrl-Left': 'goGroupLeft',
  'Ctrl-Right': 'goGroupRight',
  'Alt-Left': 'goGroupLeft',
  'Alt-Right': 'goGroupRight',
  'Ctrl-Backspace': 'delGroupBefore',
  'Ctrl-Delete': 'delGroupAfter',
  'Ctrl-S': 'save',
  'Ctrl-F': 'find',
  'Ctrl-G': 'findNext',
  'Shift-Ctrl-G': 'findPrev',
  'Shift-Ctrl-F': 'replace',
  'Shift-Ctrl-R': 'replaceAll',
  'Ctrl-[': 'indentLess',
  'Ctrl-]': 'indentMore',
  'Ctrl-U': 'undoSelection',
  'Shift-Ctrl-U': 'redoSelection',
  'Alt-U': 'redoSelection',
  fallthrough: [ 'basic', 'emacsy' ],
  'Cmd-A': 'selectAll',
  'Cmd-D': 'deleteLine',
  'Cmd-Z': 'undo',
  'Shift-Cmd-Z': 'redo',
  'Cmd-Y': 'redo',
  'Cmd-Home': 'goDocStart',
  'Cmd-Up': 'goDocStart',
  'Cmd-End': 'goDocEnd',
  'Cmd-Down': 'goDocEnd',
  'Cmd-Left': 'goLineLeft',
  'Cmd-Right': 'goLineRight',
  'Alt-Backspace': 'delGroupBefore',
  'Ctrl-Alt-Backspace': 'delGroupAfter',
  'Alt-Delete': 'delGroupAfter',
  'Cmd-S': 'save',
  'Cmd-F': 'find',
  'Cmd-G': 'findNext',
  'Shift-Cmd-G': 'findPrev',
  'Cmd-Alt-F': 'replace',
  'Shift-Cmd-Alt-F': 'replaceAll',
  'Cmd-[': 'indentLess',
  'Cmd-]': 'indentMore',
  'Cmd-Backspace': 'delWrappedLineLeft',
  'Cmd-Delete': 'delWrappedLineRight',
  'Cmd-U': 'undoSelection',
  'Shift-Cmd-U': 'redoSelection' }

Default code-mirror key mappings that are disabled in ParallaxIDE for custom key-mappings.
{  'Ctrl-Up': false,
  'Cmd-Up: false,
  'Ctrl-Down': false,
  'Cmd-Down': false,
  'Tab': false,
  'Shift-Tab': false,
  'Ctrl-T': false,
  'Cmd-T': false, }

PropGit, 2015-09-23:
@urbantumbleweed - Thank you!

PropGit, 2015-09-25:
@urbantumbleweed - Ctrl-T and Cmd-T (which are in the disabled list) are not listed in the CodeMirror key-mappings.  Is that a mistake, or is there some missing information?

PropGit, 2015-09-25:
@urbantumbleweed - Ctrl-Search-Left and Ctrl-Search-Right are Chromebook shortcuts for jumping to start/end of source, but I don't see them listed.  Are these mapped directly by Chrome OS to be equivalent?  ie: Search-Left = Home (PC) and Search-Right = End (PC)?

PropGit, 2015-09-25:
@urbantumbleweed - Do you know what Esc does?  It's mapped to "singleSelection" but I can't determine what this means or does.

PropGit, 2015-09-25:
@urbantumbleweed - I have Cmd-Up and Cmd-Down (Mac) documented as being equivalent to Ctrl-Up and Ctrl-Down (PC) which map to goDocStart and goDocEnd.  However, it's listed here as being disabled.

Is it really disabled and why?

PropGit, 2015-09-25:
@urbantumbleweed - Ctrl-Left and Alt-Left (and Ctrl-Right and Alt-Right) map to the same function goGroupLeft (or goGroupRight) but in practice (at least on Windows) they do different things.

The Ctrl-Left/Ctrl-Right move the cursor by words and the Alt-Left/Alt-Right move the cursor to the beginning or ending of the source line (like Home and End do).

It's hard to take this list at face value because there seems to be hidden behavior.

PropGit, 2015-09-25:
@urbantumbleweed - By accident, I found that Shift-Delete deletes the the current source line (deleteLine), just like Ctrl-D and Cmd-D do, but this shortcut key is not listed.

PropGit, 2015-09-25:
@urbantumbleweed - I'm now sure there are missing shortcuts in this list, such as Ctrl-C, Ctrl-X, Ctrl-V.

phated, 2015-09-28:
@PropGit most of what you are listing in all the above comments are system level shortcuts and not something we have control over.

Ctrl+C - System level copy
Ctrl+X - System level cut
Ctrl+V - System level paste
Shift+Delete - Unable to duplicate on Mac, but the delete is more like backspace (deletes one character).  Have a feeling that Windows Shift+Delete is a system thing.
Alt+Left/Alt+Right - It seems @urbantumbleweed missed the PC default for these (PC differs from mac) and they should be "goLineStart" and "goLineEnd" respectively.
Cmd+Up/Cmd+Down - There seems to be a bit of confusion, what @urbantumbleweed said is that we disabled the defaults to map them and Ctrl+Up/Ctrl+Down to our scrolling behavior (that is what the "disabled" message is about.
Esc - No idea what that means, doesn't seem to do anything.  I think it closes find if it is open.
Ctrl+T - This is mapped to "transposeChars" I believe, but we disabled it to add custom handling
Chromebook shortcuts - CodeMirror doesn't have any mappings for ChromeBook shortcuts, so everything is either deferred to the system level or Chrome OS is sending the browser a custom mapping of those keys to standard codes.  I'm not sure.

It seems that some stuff was missed, so I will have this reviewed further.

PropGit, 2015-09-28:
@phated Thanks for the info.  That makes sense.

Looking forward to your further review too.

phated, 2015-09-29:
@PropGit I went to the CodeMirror.net homepage and used my chrome developer tools to inspect the keymappings of what I believe is a default instance of CodeMirror.  I've put everything I've found in the table below with each section headed by what CodeMirror labels each.  As you can see, `Ctrl+T` exists in this table, so I think this should be everything that isn't handled by the OS.  Please reopen if this is insufficient.

| Key Command | Default Behavior |
|-------------|------------------|
| __Basic__ |
| Left | goCharLeft |
| Right | goCharRight |
| Up | goLineUp |
| Down | goLineDown |
| End | goLineEnd |
| Home | goLineStartSmart |
| PageUp | goPageUp |
| PageDown | goPageDown |
| Delete | delCharAfter |
| Backspace | delCharBefore |
| Shift-Backspace | delCharBefore |
| Tab | defaultTab |
| Shift-Tab | indentAuto |
| Enter | newlineAndIndent |
| Insert | toggleOverwrite |
| Esc | singleSelection |
| __Default__ |
| Cmd-A | selectAll |
| Cmd-D | deleteLine |
| Cmd-Z | undo |
| Shift-Cmd-Z | redo |
| Cmd-Y | redo |
| Cmd-Home | goDocStart |
| Cmd-Up | goDocStart |
| Cmd-End | goDocEnd |
| Cmd-Down | goDocEnd |
| Alt-Left | goGroupLeft |
| Alt-Right | goGroupRight |
| Cmd-Left | goLineLeft |
| Cmd-Right | goLineRight |
| Alt-Backspace | delGroupBefore |
| Ctrl-Alt-Backspace | delGroupAfter |
| Alt-Delete | delGroupAfter |
| Cmd-S | save |
| Cmd-F | find |
| Cmd-G | findNext |
| Shift-Cmd-G | findPrev |
| Cmd-Alt-F | replace |
| Shift-Cmd-Alt-F | replaceAll |
| Cmd-[ | indentLess |
| Cmd-] | indentMore |
| Cmd-Backspace | delWrappedLineLeft |
| Cmd-Delete | delWrappedLineRight |
| Cmd-U | undoSelection |
| Shift-Cmd-U | redoSelection |
| Ctrl-Up | goDocStart |
| Ctrl-Down | goDocEnd |
| __Emacsy__ |
| Ctrl-F | goCharRight |
| Ctrl-B | goCharLeft |
| Ctrl-P | goLineUp |
| Ctrl-N | goLineDown |
| Alt-F | goWordRight |
| Alt-B | goWordLeft |
| Ctrl-A | goLineStart |
| Ctrl-E | goLineEnd |
| Ctrl-V | goPageDown |
| Shift-Ctrl-V | goPageUp |
| Ctrl-D | delCharAfter |
| Ctrl-H | delCharBefore |
| Alt-D | delWordAfter |
| Alt-Backspace | delWordBefore |
| Ctrl-K | killLine |
| Ctrl-T | transposeChars |
| __PC Default__ |
| Ctrl-A | selectAll |
| Ctrl-D | deleteLine |
| Ctrl-Z | undo |
| Shift-Ctrl-Z | redo |
| Ctrl-Y | redo |
| Ctrl-Home | goDocStart |
| Ctrl-End | goDocEnd |
| Ctrl-Up | goLineUp |
| Ctrl-Down | goLineDown |
| Ctrl-Left | goGroupLeft |
| Ctrl-Right | goGroupRight |
| Alt-Left | goLineStart |
| Alt-Right | goLineEnd |
| Ctrl-Backspace | delGroupBefore |
| Ctrl-Delete | delGroupAfter |
| Ctrl-S | save |
| Ctrl-F | find |
| Ctrl-G | findNext |
| Shift-Ctrl-G | findPrev |
| Shift-Ctrl-F | replace |
| Shift-Ctrl-R | replaceAll |
| Ctrl-[ | indentLess |
| Ctrl-] | indentMore |
| Ctrl-U | undoSelection |
| Shift-Ctrl-U | redoSelection |
| Alt-U | redoSelection |
| __Mac Default__ |
| Cmd-A | selectAll |
| Cmd-D | deleteLine |
| Cmd-Z | undo |
| Shift-Cmd-Z | redo |
| Cmd-Y | redo |
| Cmd-Home | goDocStart |
| Cmd-Up | goDocStart |
| Cmd-End | goDocEnd |
| Cmd-Down | goDocEnd |
| Alt-Left | goGroupLeft |
| Alt-Right | goGroupRight |
| Cmd-Left | goLineLeft |
| Cmd-Right | goLineRight |
| Alt-Backspace | delGroupBefore |
| Ctrl-Alt-Backspace | delGroupAfter |
| Alt-Delete | delGroupAfter |
| Cmd-S | save |
| Cmd-F | find |
| Cmd-G | findNext |
| Shift-Cmd-G | findPrev |
| Cmd-Alt-F | replace |
| Shift-Cmd-Alt-F | replaceAll |
| Cmd-[ | indentLess |
| Cmd-] | indentMore |
| Cmd-Backspace | delWrappedLineLeft |
| Cmd-Delete | delWrappedLineRight |
| Cmd-U | undoSelection |
| Shift-Cmd-U | redoSelection |
| Ctrl-Up | goDocStart |
| Ctrl-Down | goDocEnd |

closed 2015-09-29 by phated

q2apro, 2017-09-20:
PS: Version 5.3 is coming out, Sep. 2017, and there is still no shortcut for duplicateLine? 

But `deleteLine` is implemented for ages :/ 

Here are all recent core shortcuts: https://github.com/codemirror/CodeMirror/blob/dda5d7220f0e3cbf8c203223ee4e7dc2d5c04bec/src/input/keymap.js