parallaxinc/Parallax-IDE #279: Are syncFileSystem file operations atomic?

PropGit, 2015-09-24:
_[syncFileSystem questions related to issue #238]_

For any individual file, does syncFileSystem treat it atomically?
  * When two devices write to the same file simultaneously, the system preempts one of them until the other is completely done with the write operation?
  * When one device writes to a file simultaneous with another device reading from that file, the system ensures the reader gets either _all_ the old content, or _all_ the new content, and _never_ a mixture of each?

Does the syncFileSystem API expose any atomic operation settings or functions to assist the caller?
  * To update a single file?
  * To update multiple files as an atomic entity?

urbantumbleweed, 2015-10-07:
The atomicity question doesn't really apply to syncFileSystem and there is no atomic file operation setting because we are not writing to Google Drive directly. All file operations are occurring on the local file system and are synced by syncFileSystem if the the file status becomes 'pending'. It is only possible to have one instance of the application open at any one time on a host so the atomicity questions for the local file system doesn't really apply there.

A file reading from syncFileSystem will get the latest version available on syncFileSystem but this may or may not be the latest version of the file (depends on Google).  However, the file will either be completely fresh or completely stale.

syncFileSystem is currently handling conflicts via 'last-write-wins'.  If both clients saved a file locally at exactly the same time, and the content is different then the there is a possibility that one of the files will become detached and the other synced.

closed 2015-10-07 by urbantumbleweed

PropGit, 2015-10-08:
>It is only possible to have one instance of the application open at any one time on a host so the atomicity questions for the local file system doesn't really apply there.

But if there's another host with the app open, atomicity does indeed apply; thus the reason for the question.  I'm forward-thinking about uses and problems.

>However, the file will either be completely fresh or completely stale.

That answers part of my question; the write operation is atomic (as I expected it would be).

>If both clients saved a file locally at exactly the same time, and the content is different then the there is a possibility that one of the files will become detached and the other synced.

In practice, I've found they each client doesn't have to write the files locally in any related time... same time doesn't matter, totally separate times cause the same problems.