parallaxinc/Parallax-IDE #288: Test updated break functionality on Windows

2fast2fourier, 2015-09-29:
I've merged the changes that use `setBreak`/`clearBreak` APIs in #287. It runs fine on the windows systems I've tried, with the caveat that any program must have a 200ms delay before writing to serial port due to framing issues covered in #194.

I'm opening this issue for @PropGit to verify these changes on his system as well. Please let me know if you run into any issues with the new build.

PropGit, 2015-09-30:
@2fast2fourier - I'm not having success. :-(

  * __I rebuilt__
    * On my Ubuntu VM, I first tried ```git pull origin master``` but it failed with a merge conflict in package.json.  I'm not sure why it didn't just fast forward.
    * So I deleted my whole repo folder and did a ```git clone git@github.com:parallaxinc/Parallax-IDE```
    * Followed by ```cd Parallax-IDE```
    * Then ```npm install```
    * Then ```npm run build```
    * Then I copied assets, fonts, icons, _locales, background.js, bundle.js, index.html, and manifest.json to my Win 7 64-bit host
    * Then I loaded it as an unpacked extension
  * __Verified I'm running Chrome v45__
[image]
  * __Ran app and tried Download__
    * I tried downloading the following code:
```
'{$STAMP BS2}
'{$PBASIC 2.5}

STOP
```

  * __and got no response__

[image]

  * __So I tried downloading the same exact code on the BSE and it was instantly successful.__

  * __BSE Reference__
--zoomed out--
[image]
-- and a zoom-in of the start of the second burst (successful download after successful identify)
[image]

  * __Parallax IDE Capture__
--zoomed out--
[image]
-- and a zoom-in of the start of the first burst (failed identify)
[image]

On the plus side, I see that there is now just a single DTR pulse at the start of every attempt, just like BSE; fantastic!

On the negative side, The Host TX seems to be holding the line in BREAK (High) constantly and then unBREAKing at the moment it should actually be BREAKing the TX.  It's the opposite of what it should be.

How does this look on your side?  You said you've had success on your Windows systems.
  * What version(s) of Windows did you use?
  * What version of Chrome are you using that/those systems?

PropGit, 2015-09-30:
@2fast2fourier - Related to our meeting discussion about the port being left in BREAK from the previous open.  I didn't know that was even possible.  Please update the code to always ensure BREAK is disengaged upon port opening so the inverted signal state is not a problem on any future port openings that the app attempts.

2fast2fourier, 2015-09-30:
@PropGit I have the changes we discussed on the call available on the `break-compat` branch. If you could check out that branch, delete the `node_modules` folder, and re-install, it should get the right dependencies.

Please let me know what you see, and post logic scans if you can. Thanks!

PropGit, 2015-10-01:
@2fast2fourier - Yes, I'll be able to check this out later tonight.  Thanks!

PropGit, 2015-10-01:
@2fast2fourier - Success!

At first I thought it failed again, but then I realized I didn't have by BOE powered- so what it was telling me was truthful.

In case it matters, I noticed that the Tx is set in BREAK the very first time the port is open, and then the app seems to clear it.

Here is the results of an identify, the first time the port is opened from the App (and happens to be with the BOE not powered).
[image]

Here's the second time (still with the BOE unpowered).
[image]
Notice the difference at the start?  I don't think this is going to cause any problems; just wanted to mention it in case it sparks a thought.

Here's the trace when the BOE is powered and Parallax IDE successfully identifies and then downloads to the BS2.
[image]

Here's the zoomed-in view of the download portion of the trace.
[image]

So, it looks like it's working well now!  I'll keep testing and report anything I find that's a problem.  Otherwise, I'll report total success again.

PropGit, 2015-10-01:
@2fast2fourier - Can you test against Chrome 45?

I'm asking because it appears you've drastically improved the situation; dare I say, completely fixed it?

In my tests I haven't had it fail to download again despite various DEBUG statements in previous downloads that are outputting version lengths and types of data.

For example, this code:
```
'{$STAMP BS2}
'{$PBASIC 2.5}

DO
  DEBUG "This is a big test where lots of text is being out plus a number: ", DEC W0, CR
  W0 = W0 + 1
LOOP
```
causes v0.8.1 to fail; however, this new version (built against the _break-compat_ branch, commit 7166893... ) hasn't failed to download against it yet!

phated, 2015-10-01:
@PropGit Great news.  He tested against Chrome 45 while putting together the fix.  I think the download failures are still possible, just much less likely.  I will get this change propagated through the dependencies and into master before release.

phated, 2015-10-01:
I've updated the dependency in master and will be removing the branches shortly

PropGit, 2015-10-01:
Sounds great.  Looking forward to testing all the other changes when you release.

PropGit, 2015-10-05:
Done.  Things are _so_ much better than before!  See Issue #194 and Issue #144.

closed 2015-10-05 by PropGit