parallaxinc/Parallax-IDE #289: Dynamically mark version in UI based on Test vs. Public channel release in CWS

PropGit, 2015-09-30:
@phated - Can the app see it's App ID?  If so, we can dynamically change something in the UI based on its App ID.

If the app determines that its App ID _is not_ the ID from our official public release CWS channel, then we'd like to mark the UI somehow to indicate that it's a release intended for testing purposes only.

Once we have the UI designed to allow selection of an "About" menu item, then we'd use this feature to decorate the version number; for example, "v1.1" would become something like "v1.1 (Test)" if the App IDE is not recognized as being from the official public CWS channel.

phated, 2015-09-30:
The application ID does seem to be exposed as `chrome.runtime.id` and we could do a string compare on that and add some visual element to the UI.  Perhaps adjust the parallax logo to include (alpha) or some other solution

PropGit, 2015-09-30:
Great!  Thanks @phated.  I'm marking this UI and we'll address it in that part of the project.