parallaxinc/Parallax-IDE #291: Test and Enhanced to Identify and Download to BS2e, BS2sx, BS2p24, BS2p40, and BS2pe

PropGit, 2015-10-06:
Verify that the app can identify and download to BS2e, BS2sx, BS2p24, BS2p40, and BS2pe modules.

At this point in time, it appears that this functionality is already built-in, or is close to being complete for Identification.

NOTE: There's some additional download considerations for these modules that will be address by other issues.

2fast2fourier, 2015-10-13:
@PropGit I'm working through these modules now, and encountered something odd with BS2e. According to the BASIC Stamp Programming Protocol 1.1 docs, BS2e identification should be a sequence of:
- Send `e` (`0x65`)
- Receive echo `e`
- Receive version byte, where `0x65`is 1.0. No other versions are mentioned.

This BS2e is Rev E, but I'm seeing a `0x67` response for the version step. Is this just a newer revision that isn't documented here?

PropGit, 2015-10-13:
@2fast2fourier - The hardware revision doesn't relate in any direct way to the firmware version; so just ignore the "Rev E" you see on the hardware.

The firmware of many modules has been revised since that document was written.  Here's a current list of expected firmware versions (but _every_ version from v1.0 through the listed version should be supported by the app; ie: identify and download to all old versions too).  Not of the versions changes the download protocol of that module.

BS2 - v1.0
BS2e - v1.2
BS2sx - v1.3
BS2p24/40 - v1.7
BS2pe24 - v1.4

The 0x67 (character 'g') represent's firmware version 1.2 for the BS2e.

PropGit, 2015-10-14:
_[CORRECTION]_

All versions 1.0 through the latest listed in the previous post should be identifiable and downloadable from the app with the exception of the BS2p... that should only support v1.2 through v1.7.  It's not the way I'd want it, but it's the way it is.

closed 2015-10-16 by phated

PropGit, 2015-10-21:
Verified working.

PR #301 (2015-10-16): Implement echo toggle functionality
#### What's this PR do?
This PR adds an Echo On/Off status and toggle at the bottom of the screen. Clicking the Echo status flips the feature on/off.
#### What are the important parts of the code?
Echo state is managed in consoleStore and maintained in the RX/TX plugin.
#### How should this be tested by the reviewer?
Connect to a board, use the transmit terminal, flip echo on/off.
#### What are the relevant tickets?
Closes #185
Closes #291 (Merged dependencies from irkenjs/bs2-programmer/pull/18)