parallaxinc/Parallax-IDE #295: Ensure "Examples" project exists on session start

PropGit, 2015-10-13:
[This issue spawns from issue #133]

App must perform the following:
  * Upon every session start (after the ["New Version"](https://github.com/parallaxinc/Parallax-IDE/issues/215) notice, if any) check for the existence of a project named "Examples"
    * Exists?  Done!
    * Doesn't exist?
      * Create a project called "Examples"
      * Fill the new Examples project with [BASIC Stamp Source Code Examples](https://drive.google.com/file/d/0B0tev9Y3WaL4Y0JkSVZZRDhVUFE/view?usp=sharing)
      * Prompt user with the message "Explore the 'Examples' project to view source code examples."
        * The prompt should have an Ok button that the user can click to hide the prompt

urbantumbleweed, 2015-10-15:
It is great that we added this issue to the project and I saw it was added to sprint 12.  As it stands we are not able to get to it this sprint.  I will mark it as deferred and then we can talk about it during the demo meeting.

PropGit, 2015-10-26:
@urbantumbleweed - I updated the examples files in the link in the first post.  The archive is now dated 2015-10-26.

phated, 2015-10-30:
It seems the discussion during sprint planning wasn't captured here.  We had decided that the default project would be the "Examples" project instead of "new project", so it doesn't seem necessary to have yet another popup dialog.

Also, we are restoring any example file that doesn't exist (deleted or newly added in a future release) instead of trying to determine if an example directory existed at the start (which wouldn't handle added files).

PR #315 (2015-10-30): Create and update example project on launch
#### What's this PR do?
This PR adds an examples folder with source files on launch. This will automatically add new files if they exist in the examples directory but are not in the user's example project, allowing new files to be added later.
#### How should this be tested by the reviewer?
Launch the app and verify that the Examples project exists and is populated. Add a new example and rebuild, verify that it is added after the fact.
#### What are the relevant tickets?
Closes #295
closed 2015-10-30 by phated